LDmicro/LDmicro #18: LDmicro for ESP8266

ioeverything, 2016-04-22:
develop ldmicro for esp8266. It will be so great work to compile ladder logic in esp8266

LDmicroGitHub, 2016-04-22:
You can use ESP8266 as UART-WIFI bridge, when the module is based on the ESP8266 is connected to the existing solutions on the basis of any other microcontroller and controlled by AT-commands, providing communication solutions to the Wi-Fi infrastructure.

LDmicroGitHub, 2016-04-22:
ESP8266 32-bit Tensilica L106-Micro Processor not in plan.

LDmicroGitHub, 2016-04-22:
@ioeverything 
You can use ESP8266 as UART-WIFI bridge, when the module is based on the ESP8266 is connected to the existing solutions on the basis of any other microcontroller and controlled by AT-commands, providing communication solutions to the Wi-Fi infrastructure.

ESP8266 32-bit Tensilica L106-Micro Processor not in LDmicroGitHub plan.

closed 2016-04-22 by LDmicroGitHub