LDmicro/LDmicro #56: Bug Report - Unclosed Rung - LDmicro Release 4.3.8, built 22:38:10 Feb 8 2018

68cme, 2018-03-28:
When editing a simple rung, including an Input Contact and a CTC, the rung remains unclosed on the screen. See 'My Screen Capture.png' in attachment.
It seems that this is just a visual issue. For example, the simulation works fine.
The export in simple text format looks like good too.

I am expecting the rung is presented closed on the screen. See 'Expected Result.jpeg' in attachment.

I am testing LDmicro 4.3.8 under W7 Pro, on a 64 bits machine. Regarding that the 'Manual' file and the 'LDmicro Tutorial' file as well, are displaying correct look, I do believe it was fine with previous versions.

68cme

PS: Thanks a lot for providing this free software. I am using ladder diagrams in professional context, and feel very, very happy to have LDmicro for private use. Great job, really.

[image]

[image]

LDmicroGitHub, 2018-05-12:
I do not like rung 2. It looks as a "short circuit".
[image]
[ldmicro_experimental.zip](https://github.com/LDmicro/LDmicro/files/1997916/ldmicro_experimental.zip)

MGPelec, 2018-05-15:
I have tried the newer version with an existing program and it works in the simulator.
The counters remain on the left and you can place an input contact without anything behind.

For me, the previous version was better for when you make mistakes, but everyone has their preferences.
Ihor it's up tp you to decide ;-)

[image]

LDmicroGitHub, 2018-05-17:
to 68cme
Please test 
ldmicro_experimental.zip