LDmicro/LDmicro #62: Good morning, this program in ldmicro works in simulation but it gives error (int-2001) when compiling for arduino, is it wrong how to do the program?

jafobrito, 2018-04-06:
[Speed.txt](https://github.com/LDmicro/LDmicro/files/1885005/Speed.txt) Good morning, this program in ldmicro works in simulation but it gives error (int-2001) when compiling for arduino, is it wrong how to do the program?

**Do you want to request a *feature* or report a *bug*?**

**What is the actual behavior?**

**If the actual behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem via Google Drive(https://drive.google.com) or similar.**

**What is the expected behavior?**

**Which versions of LDmicro, and which OS are affected by this issue? Did this work in previous versions of LDmicro?**

jafobrito, 2018-04-06:
Olá, o reset do PWM só acontece na transição de Xon para 1 e para 0, ou seja com duração de um tempo de ciclo mas mantendo ativado o PWM correspondente, não testei num Pic mas fiz compilação com os pinos assinalados para arduino (Atmega 328) e o erro que dá é na compilação, o que eu pretendo é ter uma saída de pulsos em que seja possível mudar a frequência de 500Hz para 1Khz na mesma saída (PWM), se ouver outra forma agradeço ajuda.

slyshykO, 2018-04-07:
Sorry guys, but we have serious problems with PWM when compiling to C/Arduino.
It is an LDmicro bug.

jafobrito, 2018-04-09:
Can someone explain to me how the pulse generator works in the new ldmicro,
if you can provide some examples would be better, thanks.

LDmicroGitHub, 2018-05-11:
[speed.zip](https://github.com/LDmicro/LDmicro/files/1996913/speed.zip)
Will be fixed in Release 4.3.9.2

LDmicroGitHub, 2018-05-12:
Please test with release 4.3.9.2 and close the issue if it is fixed.

closed 2018-05-13 by jafobrito