LDmicro/LDmicro #66: LDmicro not responding when compiling

parksplit, 2018-04-20:
When compiling ladder from  : 

http://cq.cx/ladder-forum.pl?action=viewthread&parent=7018&tt=1519428516

Called : "pwmexmpl_V2_with comment.ld"

Ldmicro stops responding when compiling with 328 MCU.
[image]

[image]

This is the code image:

[image]

slyshykO, 2018-04-25:
@MGPelec , may you check English variant of LDmicro?

MGPelec, 2018-04-25:
I always use the English variant of LDmicro.

Gtz
MGP

LDmicroGitHub, 2018-05-12:
Please test with release 4.3.9.2 and close the issue if it is fixed.

MGPelec, 2018-05-13:
But....how do I close this issue

slyshykO, 2018-05-13:
@MGPelec , you can't. It should be starter or owner of the repository.

MGPelec, 2018-05-13:
I received a message to close this issue after testing and I thought.... but OK

slyshykO, 2018-05-13:
@MGPelec , all participants in this conversation are receiving a notification.
If you no need of notification, you can unsubscribe.

MGPelec, 2018-05-13:
It's ok I leave it so, now that I know it ... thanks for the information

closed 2018-05-15 by LDmicroGitHub