LDmicro/LDmicro #72: When compiling ansic for arduino using REGISTER mcu, generated code includes "#", arduino doesn´t compile

parksplit, 2018-04-26:
when compiling sketch for arduino and using REGISTER mcu for disabling pull up resistor, generated code contains for example "#0x55" , this symbol "#" doesn{t let arduino compile unless you delete it see image:

[image]

LDmicroGitHub, 2018-04-26:
== Release 4.3.8.7
    * Partially fixed: When compiling ansic for arduino using REGISTER mcu,
    generated code includes "#", arduino does not compile (#72)

TODO: in many other places :(
NOTE: Do not close until full fixed.

LDmicroGitHub, 2018-05-12:
Please test with release 4.3.9.2 and close the issue if it is fixed.

closed 2018-05-15 by LDmicroGitHub