LDmicro/LDmicro #82: problem "persist" arduino mega

lucianod6, 2018-05-08:
I have been programming PLC for over 20 years and I can say that LDmicro is 
a program at the highest professional level, really fantastic!
i using arduino mega 2560 ..I want to report the following problem :
Persist does not work ..I have identified the problem , the generate code from ldmicro is:
    if(Read_Ib_rung_top()) {
        if(!Read_Ib_once_4_PERSIST_variabile_sp_minuti()) {
            Write_Ib_scratch(0);
            Write1_Ib_scratch(); // EEPROM is ready
            if(!Read_Ib_scratch()) {
                Write_Ib_once_4_PERSIST_variabile_sp_minuti(1);
                Ui_variabile_sp_minuti = EEPROM_read(0) + (EEPROM_read(1) << 8);
            }
        } else {
            Write_Ib_scratch(0);
            Write1_Ib_scratch(); // EEPROM is ready
            if(!Read_Ib_scratch()) {
                Ii_tmpVar24bit = EEPROM_read(0) + (EEPROM_read(1) << 8);
                if(Ii_tmpVar24bit == Ui_variabile_sp_minuti) {
                } else {
                    EEPROM_write(0, Ui_variabile_sp_minuti & 0xFF);
                    EEPROM_write(1, Ui_variabile_sp_minuti >> 8);
                }
            }
        }
    }
it does not work

I modified it as follows
    if(Read_Ib_rung_top()) {
        if(!Read_Ib_once_4_PERSIST_variabile_sp_minuti()) {
            Write_Ib_scratch(0);
           // Write1_Ib_scratch(); // EEPROM is ready
            if(!Read_Ib_scratch()) {
                Write_Ib_once_4_PERSIST_variabile_sp_minuti(1);
                Ui_variabile_sp_minuti = EEPROM_read(0) + (EEPROM_read(1) << 8);
            }
        } else {
            Write_Ib_scratch(0);
           // Write1_Ib_scratch(); // EEPROM is ready
            if(!Read_Ib_scratch()) {
                Ii_tmpVar24bit = EEPROM_read(0) + (EEPROM_read(1) << 8);
                if(Ii_tmpVar24bit == Ui_variabile_sp_minuti) {
                } else {
                    EEPROM_write(0, Ui_variabile_sp_minuti & 0xFF);
                    EEPROM_write(1, Ui_variabile_sp_minuti >> 8);
                }
            }
        }
    }
it works properly
Release 4.3.9.0

LDmicroGitHub, 2018-05-11:
Fixed in 4.3.9.2

closed 2018-05-11 by LDmicroGitHub

LDmicroGitHub, 2018-05-12:
Please test with release 4.3.9.2 and reopen the issue if it is not fixed.