LDmicro/LDmicro #89: PIC 16F628

milan1984, 2018-05-15:
**Do you want to request a *feature* or report a *bug*?** YES

**What is the actual behavior?**ERROR DURING COMPILING ''INTERNAL ERROR AT LINE 2390 FILE PIC16.CPP"

**If the actual behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem via Google Drive(https://drive.google.com) or similar.**

**What is the expected behavior?**

**Which versions of LDmicro, and which OS are affected by this issue? Did this work in previous versions of LDmicro?**VER 4.3.9.2

LDmicroGitHub, 2018-05-16:
Please attach the LD file

milan1984, 2018-05-16:
[test 16f628.zip](https://github.com/LDmicro/LDmicro/files/2010108/test.16f628.zip)