LDmicro/LDmicro #113: Page Error

deshpande67, 2018-06-06:
While Compiling a Page error is coming. 
I can't attach .ld file here so zip file is attached. 
[TEMP_4TIMER.zip](https://github.com/LDmicro/LDmicro/files/2076492/TEMP_4TIMER.zip)

parksplit, 2018-06-06:
[image]

LDmicroGitHub, 2019-02-02:
Messages clarified.  Please recheck the issue with the Release 5.3.0.0 or later.
[image]  
[image]  
[image]  
Unfortunately, LD and HEX are too big for PIC16F88.