LDmicro/LDmicro #130: ADC not sent to USART

MoinRajpoot, 2018-08-07:
**Do you want to request a *feature* or report a *bug*?**

*can any one help me why UART does not sending variable data from ADC 
"
[image]
[image]

slyshykO, 2018-08-07:
Please, attach *.ld script. (as zip archive)

LDmicroGitHub, 2018-08-08:
[image]  
[image]

MoinRajpoot, 2018-12-24:
ok thanks

closed 2018-12-24 by MoinRajpoot