LDmicro/LDmicro #149: LCD with Arduino UNO

MoinRajpoot, 2018-12-24:
can we use LCD with LDmicro plz need help i wanna build a project that require LCD for data entry display

TecnoCalubert, 2018-12-24:
Si mírate el Foro

MoinRajpoot, 2019-08-05:
dear TecnoCalubert  will you teach me LDmicro ?????

TecnoCalubert, 2019-08-05:
que necesitas.
what do you need

TecnoCalubert, 2019-08-05:
what do you need

MoinRajpoot, 2019-08-05:
Dear I want to make a pressure control system that contains 2 pumps 1
pressure sensor
LCD used to monitor pressure in vessel
Used to set require pressure and monitor pump on off status the whole
system run from 9am to 5 pm
I can make whole program but can't setup LCD
I want your help in programming a LCD

TecnoCalubert, 2019-08-05:
Sabes algo de programación de LDmicro, hablas español y dime que idiomas hablas y lo lees.
You know something about LDmicro programming, you speak Spanish and tell me what languages you speak and read it

MoinRajpoot, 2019-08-05:
English

TecnoCalubert, 2019-08-05:
pasarme tu es calera y te hago el LCD
pass me your ladder and I make you the LCD