LDmicro/LDmicro #168: LDmicro 5301 does not accept MCU crystal <1MHz

LDmicroGitHub, 2019-04-01:
http://cq.cx/ladder-forum.pl?action=viewthread&parent=8367&tt=1554155021

MGPelec, 2019-04-02:
LDmicro 5301
We can not use MCU frequency's below 1MHz, it generate an error.
The internal oscillator of 500kHz is therefore not available.
Thx.

[image]

TecnoCalubert, 2019-04-02:
El problema es que colocan (.) Cuando se tiene que colocar (, ) No hay ningun problema.
The problem is that they place (.) When it has to be placed (,) There is no problem.

MGPelec, 2019-04-02:
I can not place a comma (,) ldmicro does not accept a (, ) only a (.)

Grtz

MGPelec, 2019-04-02:
It works in version 4430, so I think it's a bug.

TecnoCalubert, 2019-04-02:
No entendí muy bien, este tema hace bastante tiempo ya se hablo de las comas y puntos y se cerro el caso

MGPelec, 2019-04-02:
Sorry, my fault, it's no bug but my lack of knowledge of windows.
Problem solved. thx to Ihor for the  support.

I do not know if I can close this thread and how....

LDmicroGitHub, 2019-04-02:
What value used as a "Decimal symbol" in your Windows?   
Try to change it.  
[image]

closed 2019-04-02 by LDmicroGitHub