LDmicro/LDmicro #171: UART simulator does not work correctly.

MGPelec, 2019-04-08:
There is probably a bug in the UART simulator of V5302.
If I simulate and I enter a char through the simulator, then the program freezes.

[test-UART-sim.zip](https://github.com/LDmicro/LDmicro/files/3054166/test-UART-sim.zip)

TecnoCalubert, 2019-04-08:
Yo e probado en la versión \build5302\ldmicro-es.exe.
y ami me funciona correcto .

MGPelec, 2019-04-08:
I also tried your LDmicro-es version and it doesn't work either.

TecnoCalubert, 2019-04-08:
Adjunto winrar, te e hecho para que veas como esta en LDmicro, Creo
que esta mal traducido el tx con rx de enviar y recivir, puedes probarlo
al contrario 

> I also tried
your LDmicro-es version and it doesn't work either. 
> 
> You are
receiving this because you commented.
view it on GitHub [1], or mute the thread [2].

-- 

Tecno Calubert SL.
(Calubert) Paulino/Gabriel Cano
Fijo 0034-931188277 Movil 0034-639335284

E-Mail calubert@tecnocalubert.com 
Web tecnocalubert.com

AVISO LEGAL

Este mensaje se dirige exclusivamente a su destinatario y puede
contener información privilegiada o confidencial. Si no es Vd. El
destinatario indicado, queda informado de que la utilización,
divulgación y/o sin autorización está prohibida en virtud de la
legislación vigente. Si ha recibido este mensaje por error, le rogamos
que nos lo comunique inmediatamente por esta misma vía
baja@tecnocalubert.com y proceda a su destrucción. 
Si no desean recibir
información de Tecno Calubert sl. baja@tecnocalubert.com

P Antes de
imprimir este e-mail piense bien si es necesario hacerlo

cabecera

Links:
------
[1]
https://github.com/LDmicro/LDmicro/issues/171#issuecomment-480814133
[2]
https://github.com/

TecnoCalubert, 2019-04-08:
Hola 
creo que esta creo que esta mal el revivir con el transmitir
[test-UART-sim2.txt](https://github.com/LDmicro/LDmicro/files/3054564/test-UART-sim2.txt)

TecnoCalubert, 2019-04-08:
no tenia el win, zip Te lo e puesto para que veas, la modificación es para yo poder simularla 
[test-UART-sim2.zip](https://github.com/LDmicro/LDmicro/files/3054607/test-UART-sim2.zip)

TecnoCalubert, 2019-04-08:
dime si lo entendiste

MGPelec, 2019-04-08:
I wonder if you know what I mean, your programs doesn't work in the simulator either.

TecnoCalubert, 2019-04-08:
Pasarme tu Email i te explicare ya que no quier cargar el foro con comentarios que no son reales hasta describir el caso calubert@tecnocalubert.com

TecnoCalubert, 2019-04-08:
Si lo se en mi win-10-pro es correcto con los programa que e cargado funciona en mi simulador, yo lo e probado antes de envíalo

MGPelec, 2019-04-08:
If that is the case I will wait until Ihor tested it (or someone else)
In all other versions of LDmicro it's working, not in V5xxx

Fault: I can't send a char from the keyboard to UART in  the simulator.

TecnoCalubert, 2019-04-08:
OK

parksplit, 2019-04-08:
yeah, it freezes

TecnoCalubert, 2019-04-09:
Por Favor decir si es correcto
Please tell if it is correct
[uart Prueba.zip](https://github.com/LDmicro/LDmicro/files/3058012/uart.Prueba.zip)
[uart.log](https://github.com/LDmicro/LDmicro/files/3058013/uart.log)

Abrir programa, una vez abierto,  simular, una vez esta simulando, bajar a bajo  del ld-micro,
Clicas con doble clic, donde pone (outch  UART rx ) y se abrirá una ventana en esa ventana colocar en la línea superior un 1, la línea inferior dejarla igual (esta línea es para decirle cuantos dígitos queremos, la lectura y escritura la hace en Nº  exadecimal.
Nota: al comienzo de simular ya comenzara a ver el simulador terminal UART.

Open program, once opened, simulate, once it is simulating, go down to the low-micro,
Clicks with double click, where it puts (outch UART rx) and a window will open in that window place on the top line a 1, the bottom line leave it the same (this line is to tell you how many digits we want, reading and writing it in Exact number
Note: at the beginning of simulation you will start to see the UART terminal simulator

MGPelec, 2019-04-09:
Calubert, you still do not understand it.

In the simulator you can simulate sending a char with your keyboard.
Open my first program, simulate it, then you see a second UART window, click on that window to activate that window and push a key, then your program freezes.
No need to send more programs, it does not work in V5xxx I'm using V4430 now for simulating en V5xxx to work with.

TecnoCalubert, 2019-04-09:
https://github.com/LDmicro/LDmicro/issues/148

TecnoCalubert, 2019-04-09:
lo entendí en su momento, por eso digo que no hay errores  a mi me funciona tu Programa ldmicro con el teclado. Uso win10 Pro 64 bit y 32bit, mira tu ordenador que este bien actualizado para programas 32bit y 64 bit el ordenador,  ya que observemos varios tu fallo de punto con la coma

TecnoCalubert, 2019-04-09:
Para mi el caso #171 esta cerrado

MGPelec, 2019-04-09:
Not for me and I'm not the only one...
My computer is Win7 Ultimate 64 and uptodate.