LDmicro/LDmicro #174: MSVCP100.dll is missing

Davi767, 2019-04-13:
ALL the LDMicro versions after the 4.4.3.0 does not start because MSVCP100.dll is missing.

Is so difficult to ADD all the necessary libraries IN THE BUILD and not leave the poor user to find by herself and try to install various versions if MS Visual C++ Scrap to find the right one??

LDmicroGitHub, 2019-04-13:
Please, install [Microsoft Visual C++ redistributable](https://www.google.com/search?newwindow=1&ei=gQayXM2fIZKtrgTlgLeICw&q=Microsoft+Visual+C%2B%2B+redistributable&oq=Microsoft+Visual+C%2B%2B+redistributable&gs_l=psy-ab.3..35i39j0i203l3j0l2j0i203l2j0j0i203.10134.10947..12449...0.0..0.109.192.1j1......0....1..gws-wiz.......0i71.pYdoV2wzJIg)  
Some links...  
[The latest supported Visual C++ downloads](https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads)  
[Visual C++ Redistributable for Visual Studio 2015](https://www.microsoft.com/en-us/download/details.aspx?id=48145)

Davi767, 2019-04-15:
VC2017 doesn't fix anything.
VC2015 doesn't fix anything.
VC2013 doesn't fix anything.
VC2012 download link is broken.

I've solved anything, lost only some time.

I repeat the question: is so difficult to add ALL THE NECESSARY FILES IN THE BUILD and not leave the user to install tons of crap and pray that works? Thanks.

LDmicroGitHub, 2019-04-16:
The Microsoft Visual C++ 2010 Redistributable Package installs runtime components of Visual C++ Libraries required to run applications developed with Visual C++ on a computer that does not have Visual C++ 2010 installed.  
Please select the bit depth and language of the installed operating system:  
[Microsoft Visual C++ 2010 Redistributable Package (x86)](https://www.microsoft.com/en-US/download/details.aspx?id=5555)  
[Microsoft Visual C++ 2010 Redistributable Package (x64)](https://www.microsoft.com/en-us/download/details.aspx?id=14632)

Nenca, 2020-12-26:
olá  pessoal, bom dia  o  meu problema é quando  tento abrir a versão build5404  ou  build5405  vem o  erro  0xc000007b    já  estalei varias verções do Visual C ++ e  dll_s  e não  resolveu  então ????   grato