LDmicro/LDmicro #175: Feature request: little change for Quadrature Encoder

DanielPrjct, 2019-04-13:
Feature request: little change for Quadrature Encoder

Dear Ihor:
Could you change the type of Output Dir of Quadrature Encoder, from Y  (pin) to R (internal relay, mark)?
I.e.:
Output Dir: RqDir0

IMHO, I/O ports (pins) are some of the most valuable elements in microcontrollers (in special on PICs or on AVR, with the exception of ATmega2560 wich has lot of them).
Using an Output Pin only for signalization (increment or decrement in the internal counter variable) seems to me a waste of scarce resources.

Using an internal relay R for Output Dir in QE doesn't cause any trouble:
If user wants to manage externally that signal, simply must read the internal relay by means an open contact and to connect it to a Y (pin) coil.

```
       ROutputDir                            YDir
---------]  [--------------------------------(  )--

```

Thanks in advance!
Best regards,
DanielH

LDmicroGitHub, 2019-04-14:
Output Dir is optional and can be empty.

LDmicroGitHub, 2019-04-14:
Fixed: Quadrature Encoder, Issue  #175  
Output Dir can be Y (pin) or R (internal relay).

closed 2021-03-20 by LDmicroGitHub