LDmicro/LDmicro #197: sink and source

MoinRajpoot, 2019-08-29:
can we use arduino  with sink and source options 
for input and output as normally arduino interface in market use sink (low level trigger) option

TecnoCalubert, 2019-08-29:
I do not understand your question
No entiendo bien tu pregunta

MoinRajpoot, 2019-08-29:
Can we use Ld micro to program a atmega 328 with low level input trigger
and low level output trigger

TecnoCalubert, 2019-08-29:
Yes LDMicro The internal resistors of the Atmega328 can be selected
Si LDMicro Se puede seleccionar las resistencias internas del Atemega328

MoinRajpoot, 2019-08-29:
How can we do that

TecnoCalubert, 2019-08-29:
There I set the example for a doubt look at it
http://cq.cx/ladder-forum.pl?action=viewthread&parent=8955&tt=1566986668

MoinRajpoot, 2019-08-29:
OK thanks dear

TecnoCalubert, 2019-08-29:
Tell me if you understood
Dime si lo entendiste

LDmicroGitHub, 2021-03-07:
There is a shorter way.  
LD file at the forum https://cq.cx/ladder-forum.pl?action=viewthread&parent=8955  
[image]