LDmicro/LDmicro #217: analog input & pwm output

safit1, 2020-04-02:
hai everybody.
versions of LDmicro 5404.
I'm facing this problem. simulation work well but hardware fail to function. why.
I'm using arduino uno.

[image]
[image]

anyone can help me

LDmicroGitHub, 2020-04-07:
Please pack LD file to zip file and attach zip file here.

safit1, 2020-04-07:
[pwm-simple1.zip](https://github.com/LDmicro/LDmicro/files/4447389/pwm-simple1.zip)

safit1, 2020-04-07:
[PWM_POT1.zip](https://github.com/LDmicro/LDmicro/files/4447396/PWM_POT1.zip)

safit1, 2020-04-07:
[pwm-simple.zip](https://github.com/LDmicro/LDmicro/files/4447411/pwm-simple.zip)

this ld code work well. it seem this need version can't compile analog input but pwm output work fine.