LDmicro/LDmicro #219: Not working on Atmega8

jrelectrokits, 2020-04-07:
hi
i have write code in latest version v5.4.0.4 for atmega8 DIP and 32Pin package but both are not working on controller.

i have attached ld code, can u check it for me ?

thanks in advance.

**Desktop 
 - OS: windows 10 pro
 
[code.zip](https://github.com/LDmicro/LDmicro/files/4442529/code.zip)

TecnoCalubert, 2020-04-07:
what do you mean it doesn't work, in chic or in simulation
a que te refieres que no funciona, en el chic o en la simulación

jrelectrokits, 2020-04-07:
In simulator works perfectly, but not working in chip

LDmicroGitHub, 2020-04-07:
PB6 and PB7 are oscillator pins. Try to use PORTC pins instead. 
[atmega8 32pin.zip](https://github.com/LDmicro/LDmicro/files/4446850/atmega8.32pin.zip)

LDmicroGitHub, 2020-04-07:
You can use PB6 and PB7 as output only when set internal oscillator AVR fuses.
https://www.google.com/search?q=set+avr+fuses&tbm=isch&ved=2ahUKEwiAqvzNl9foAhVUvyoKHUDtC5oQ2-cCegQIABAA&oq=set+avr+fuses&gs_lcp=CgNpbWcQA1CCkwJYkpkCYKieAmgAcAB4AIABvwGIAdYDkgEDMy4xmAEAoAEBqgELZ3dzLXdpei1pbWc&sclient=img&ei=muaMXoD1LtT-qgHA2q_QCQ&bih=768&biw=1600

jrelectrokits, 2020-04-08:
I have remove pin 7 & 8 and trying but not working

Also not supporting atmega8 dip (28 pin) ic..