LDmicro/LDmicro #222: The setings for the Pull-up resistors are are not reset when you start a new program

GregBlackwell, 2020-06-13:
Test case to demonstrate this issue:

Start the current version of LDmicro version 5.4.0.4
Choose the microcontroller PIC16F628 
- Menu: Settings -> Microcontroller -> Microchip Pic10-16 MCUs -> Microchip PIC16F628 18-PDIP or 18-SOIC

Set the Pull-up resistors
- Menu: Settings -> Set Pull-up input resistors
- Enter a value of 0x1a for Port RA
- Enter a value of 0x2b for Port RB

Save the program
- Menu: File -> Save
- Enter a filename and save

Start a new program
- Menu: File -> New

Choose the microcontroller Atmel AVR ATmega2560 100-TQFP
- Menu: Settings -> Microcontroller -> Atmel AVR MCUs -> Atmel AVR ATmega2560 100-TQFP

Display the Pull-up resistor settings
- Menu: Settings -> Set Pull-up input resistors

Port PA has the initial value 0x1a and Port PB has the initial value 0x2b. The Pull-up resistor settings are not being reset when a new program is started.

LDmicroGitHub, 2021-03-20:
Try it in v5.4.1.1

closed 2021-03-22 by LDmicroGitHub

GregBlackwell, 2021-04-17:
This issue is fixed in v5.4.1.1

commit 36cbb0c8 (2021-03-22): Reset the Pull-up resistors when you start a new program. Fix #222