LDmicro/LDmicro #226: The C code generated for the SUBPROG instruction has multiple duplicate function definitions and the body of the function is empty

GregBlackwell, 2021-01-03:
Compiling C code for the SUBPROG instruction generates a function definition for each instance of the GOSUB instruction that refers to it. When there are multiple GOSUBs there are multiple function defintions generated whiich will cause compilation of that C program to fail.

The v5.4.x.x versions of ldmicro no longer generates the correct code in the body of the SUBPROG funcition and the body of these functions is empty.
The v4.2.2.0 version of ldmicro gererates the correct code in the body of the SUBPROG function.

Test case to demonstrate this issue:
Download the gosub.ZIP file from the wiki page "SUBPROG, ENDSUB and GOSUB instruction" and extract the file gosub.ld
- This program has two SUBPROGs Pulse1 and Pulse2. There are 4 GOSUBs to Pulse1 and 3 GOSUBs to Pulse2.

Start the current version of LDmicro version 5.4.0.5 and open the file gosub.ld

Compile AnsiC an inspect the generated c code. 
It has the following 4 duplicate function definitions for Call_SUBPROG_Rung6()  and 3 duplicates for Call_SUBPROG_Rung23() and the body of all these fuctions are empty.
void Call_SUBPROG_Rung6() { // LabelRung2
}
void Call_SUBPROG_Rung6() { // LabelRung2
}
void Call_SUBPROG_Rung6() { // LabelRung2
}
void Call_SUBPROG_Rung23() { // LabelRung1
}
void Call_SUBPROG_Rung23() { // LabelRung1
}
void Call_SUBPROG_Rung23() { // LabelRung1
}
void Call_SUBPROG_Rung6() { // LabelRung2
}

Assign the microcontroller Atmel AVR ATmega8 28-PDIp and choose pins for the unassigned pins.
Compile Atmel WinAVR C and inspect the generated c code. It has the same two issues.
Compile Atmel AVR-GCC and inspect the generated c code. It has the same two issues.

Attempting to compile the duplicate function definitions give these errors
gosub.c:133:6: error: redefinition of 'Call_SUBPROG_Rung6'
 void Call_SUBPROG_Rung6() { // LabelRung2
      ^~~~~~~~~~~~~~~~~~
gosub.c:136:6: error: redefinition of 'Call_SUBPROG_Rung6'
 void Call_SUBPROG_Rung6() { // LabelRung2
      ^~~~~~~~~~~~~~~~~~

LDmicroGitHub, 2021-03-20:
Please attach the ld file.

GregBlackwell, 2023-01-07:
The ld file gosub.ld: 
[gosub.zip](https://github.com/LDmicro/LDmicro/files/10366257/gosub.zip)