LDmicro/LDmicro #227: BuildAVR.bat

onostechnology, 2021-02-09:
**Do you want to request a *feature* or report a *bug*?**
I am reporting a bug

**What is the actual behavior?**
When the Build C Solution is selected, the following error is generated

      blink.c: In function 'mainPlc':
      blink.c:152:11: error: 'true' undeclared (first use in this function)
           while(true) {
                 ^~~~

**If the actual behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem via Google Drive(https://drive.google.com) or similar.**
https://drive.google.com/file/d/1qnf018qwz9ePYDYK4ktIK01lx3f6-hat/view?usp=sharing

**What is the expected behavior?**
This error is manually rectified by changing the "true" to figure "1",

**Which versions of LDmicro, and which OS are affected by this issue? Did this work in previous versions of LDmicro?**
Release 5.4.0.5, built 00:59:48 Oct 30 2020.

LDmicroGitHub, 2021-03-13:
Fixed in release 5.4.1.0

closed 2021-03-13 by LDmicroGitHub

onostechnology, 2021-03-14:
Hi,

Thanks for responding to the issue and for the expected new version. Just
to point out that my latest trial of BuildARM and BuildPIC also generated
similar error  I hope these get fixed too.

I am very grateful to you guys for your work on LDmicro. I wish I am an
expert in C programming language so I could join the team. But I think I
can support by testing and updating the Wiki How To page if you are ok with
it.

Best Regards
Onoriode