LDmicro/LDmicro #229: Persist instruction is not working in new release,

jrelectrokits, 2021-03-08:
**Do you want to request a *feature* or report a *bug*?**
Report a bug
**What is the actual behavior?**
When i make persist variable, it is not stored in eeprom
**If the actual behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem via Google Drive(https://drive.google.com) or similar.**

**What is the expected behavior?**

**Which versions of LDmicro, and which OS are affected by this issue? Did this work in previous versions of LDmicro?**

jrelectrokits, 2021-03-08:
If some one have sample code, then please share with me.

LDmicroGitHub, 2021-03-13:
[MAKE PERSISTENT operation](https://github.com/LDmicro/LDmicro/wiki/MAKE-PERSISTENT-operation)

LDmicroGitHub, 2021-03-13:
May you publish the ld file?

jrelectrokits, 2021-03-15:
attached ld file, please let know if there is any issue in code.
thanks

[TIMER_3DIGIT.zip](https://github.com/LDmicro/LDmicro/files/6140716/TIMER_3DIGIT.zip)

jrelectrokits, 2021-07-28:
> May you publish the ld file?

hi
any update regarding persistent instruction, it is not working in atmega8 controller.