LDmicro/LDmicro #232: LD Micro 5.4.1.1 AtMega328P (28 pin) Doesn't Analog Input A0 not available

ckiernan, 2021-08-03:
**Do you want to request a *feature* or report a *bug*?** bug

**What is the actual behavior?** Using an ATMega328P-PU (28 pin), when adding a Read ADC command to ladder code and I go to add an analog pin to my configuration, Analog input, PC0 (ADC0/PCINT8) and its not an avialable choice. Using the attached code, I can run this on a AATMega2560 but not on the ATMega328

**If the actual behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem via Google Drive(https://drive.google.com) or similar.** 
Steps to reproduce:

1. open the attached code in LDMicro and select the AVR ATMega2560 processor.
2. assign the analog input to analog input PC0 and assign the output to any output with an led wired to it. connect an analog input to PC0
2. Compile the code using the AVR-GCC method described in this video: 
[https://youtu.be/1ooMWbz8Zzs](https://youtu.be/1ooMWbz8Zzs). The code compiles and loads successfully.
3. connect a serial monitor to the uart output and see the analog value change
4. Next connect to an AVR ATMega328P-PU (ardunio uno) and attempt to assign the analog input to PC0. It does not appear in the I/O Pin list.

**What is the expected behavior?** PC0 would be available as an option in the I/O List for the ATMega328P-PU

**Which versions of LDmicro, and which OS are affected by this issue? Did this work in previous versions of LDmicro?** Version 5.4.1.1

Sample Code:
[Analog_Input.zip](https://github.com/LDmicro/LDmicro/files/6920264/Analog_Input.zip)

ckiernan, 2021-08-04:
Update, This behavior does not exist in Build 5403
[image]

biatel1, 2021-08-05:
Strange... Just started new project, put one ADC and selected ATMEGA328 28-PDIP and...
[image]
For 32 pin are available A6 and A7 as well.

biatel1, 2021-08-05:
Also your project opens with ADC0 already assigned, without any operation from my side:
[image]

BTW: The only problem I have with all 5.4.x.x versions is that I2C is not working for ATMEGA8 (both 32 and 28 pin). So I must use 5.3.x.x in this case. But generally it looks like your description i.e. pins SCL and SDA are not available while are in ATMEGA328.

ckiernan, 2021-08-07:
Yep, my mistake, I now see that it is available.

closed 2021-08-07 by ckiernan