LDmicro/LDmicro #233: In the current version 5.4.1.1 it is no longer possible to choose an MCU clock frequency less than 1MHz

GregBlackwell, 2021-12-13:
Prior to version 5.3.0.0 it was possible to choose an MCU clock frequency that was less than 1 MHz. e.g. 0.5
In the current version 5.4.1.1 choosing an MCU frequency that is less than 1 MHz generates the error "Zero crystal frequency not valid, resetting to 16MHz."

Test case to demonstrate this issue:
Start the current version 5.4.1.1 of the English version of LDmicro on a PC with an English locale.
- Menu -> Settings -> MCU Parameters
- Change the MCU Crystal Frequency (MHz) value to 0.5 and click OK
You will get the error dialog "Zero crystal frequency not valid, resetting to 16MHz."

This appears to have been introduced in v5.3.0.0 when the following code was added to the function ShowConfDialog() in the file confdialog.cpp:
        ///// Added by JG:  convert '.' to ',' for atof()
        for(size_t i = 0; i < strlen(cancel_buf); i++)
            if(cancel_buf[i] == '.')
                cancel_buf[i] = ',';
        /////

Wen I checked I found out that most other locales have a different character for the decimal separators than the period used by English locales.

I don't have a PC running a locale that uses a comma character as the decimal separator so I can't do the obvious test to see if the MCU frequency is correctly displayed with a comma as the decimal separator and if the frequency can be set to a value less than 1MHz using that separator.

2nd test case:
Start the current version 5.4.1.1 of the English version of LDmicro on a PC with an English locale.
- Menu -> Settings -> MCU Parameters
- Change the MCU Crystal Frequency (MHz) value to 0.5 and click OK
You will get the error dialog "Zero crystal frequency not valid, resetting to 16MHz."

- Menu -> Instruction -> Insert Comment
- Menu -> File -> Save As and choose the filename "test.ld"
Open the file test.ld and the frequency is correctly recorded as "CRYSTAL=500000 Hz"

biatel1, 2021-12-13:
Using 5.3.1.0 due to other problems with 5.4.1.1 and having no problem with low speec MCU.

GregBlackwell, 2021-12-14:
biatel, I just tested the v5.3.1.0 English version (Help -> About reports Release 5.3.1.0, built 19:31:41 May 30 2019.) on Win 7 with Local=English (Australian).

When I set the parameter "MCU Crystal Frequency (MHz)" from the default 16.000000 to the new value 0.5 I get the same error "Zero crystal frequency not valid, resetting to 16MHz." when I click OK for the dialog.

biatel1, 2021-12-14:
Maybe there's something inside a decimal separator. In my country it's set in Windows as comma (,) and I'm not able to write 0,25 in simply way. But I'm doing this as follows: first I overwrite the whole digits, then I overwrite the decimal places, and never touch the separator. Then it works.

GregBlackwell, 2021-12-17:
biatle1,

I'm using the "English (Australia) version of Windows 7.
Using the current English version 5.3.1.0 of LDmicro:
- When I open the menu Settings -> MCU Parameters the default value for the field "MCU Crystal Frequency (MCU)" is 16.000000
- I am able to completely delete the contents of this field. 
- I am able to enter the value 0.5.
- I do not have any of the problems that you describe setting a value less than 1.
- If the value is 0.5 then when I click the "Close" button I get the error "Zero crystal frequency not valid; resetting to 16MHz."