LDmicro/LDmicro #245: Performance issue under Windows 10

tigrouind, 2023-03-11:
**Describe the bug**
After inserting/deleting a few instructions on a LD file of a reasonable size (50 rungs, 100 contacts), LDmicro becomes slow.
Once this happen, any operation (eg: insert, edit, delete) take at least one second while normally (eg : after fresh start of program) this is instantaneous. Also when this occurs, closing LDmicro take a lot of time (20-30 seconds).

**To Reproduce**
Steps to reproduce the behavior:
1. Start LDmicro 
2. Insert some amount of contacts using C key (eg: about 50 contacts). (eg: both in parallel and in series)
3. At some point, insertion operation will become slow (deleting some contacts seems to help only slightly).

**Expected behavior**
LDmicro remain responsive and fast (most operations should be instantaneous).

**Desktop (please complete the following information):**
 - OS: Windows 10 22H2
 - Version 5.4.1.1

parksplit, 2024-08-08:
is there a way to fix this issue?

tigrouind, 2024-08-08:
If you looking for a quick fix / workaround, you could lower DISPLAY_MATRIX_X_SIZE and DISPLAY_MATRIX_Y_SIZE back to their [original values](https://github.com/LDmicro/LDmicro/blob/8fc0cc7efaf917641ff2cdcb114a7fd7847215a2/ldmicro/ldmicro.h#L589).

The size of the circuit will be limited to 16 columns and 512 rows but that should be enough for most projects (can be slightly increased if needed).

For a proper fix, investigation need to be made. First to check if bug has been introduced at some point or existed since beginning. The main issue is that ForgetFromGrid is called while the element to be freed is not more in DisplayMatrix array. Was it already like in initial commit ?

parksplit, 2024-08-09:
Hi tigrouind,
I hope this message finds you well. I wanted to ask if it would be possible to build LDmicro version 5.4.1.1 with the original values for DISPLAY_MATRIX_X_SIZE and DISPLAY_MATRIX_Y_SIZE that you proposed.
I believe reverting to those original settings might help address the performance issues we've been discussing.
Thank you for your efforts on this project, and I look forward to your response.
Best regards,
Parksplit

tigrouind, 2024-08-09:
@parksplit I have been able to fix the issue. Here a [32-bit Windows build](https://github.com/user-attachments/files/16566975/ldmicro.zip).

Lowering DISPLAY_MATRIX_X_SIZE and DISPLAY_MATRIX_Y_SIZE fix the problem (as expected). This is not, however, how I fixed it in this build. I found a better way (without lowering the limits).

This explain what is going wrong :
In [FreeEntireProgram()](https://github.com/LDmicro/LDmicro/blob/master/ldmicro/circuit.cpp#L1263) we call [ForgetEverything()](https://github.com/LDmicro/LDmicro/blob/master/ldmicro/schematic.cpp#L331) we clear out DisplayMatrix array (so it became fill with zeroes). Then right after we call [Prog.reset()](https://github.com/LDmicro/LDmicro/blob/master/ldmicro/plcprogram.cpp#L24) which will try to clearout elements from DisplayMatrix (but they have already been cleared). Because of that, it loops on the whole array for each element and it takes a really long time.

This issue seems to have been there since the [beginning](https://github.com/LDmicro/LDmicro/blob/8fc0cc7efaf917641ff2cdcb114a7fd7847215a2/ldmicro/circuit.cpp#L639). It is the increase of DISPLAY_MATRIX_SIZE constants that make the issue more obvious.

To fix this, simply comment out the for loop in [PlcProgram::reset()](https://github.com/LDmicro/LDmicro/blob/master/ldmicro/plcprogram.cpp#L26). I did not push a pull request since this project seems to be no more active since more than 3 years.

parksplit, 2024-08-10:
thank you very much for this 32 bit version, I´m going to test it.

tigrouind, 2024-08-11:
Yes, you can share it on the forum. Btw, I have created a [better version](https://github.com/user-attachments/files/16571204/ldmicro.zip).

The way it works : the X / Y position of each element in the grid is now stored in the element itself.
When an element need to be removed from the grid, there is no more need to loop on the grid to find the element. This version might be even faster than the 32 x 512, despite the grid being 256 x 9999.

For anyone interested with the changes: here is the related [commit](https://github.com/tigrouind/LDmicro/commit/6c8a1af2cc21da46bf0171d5aa05090e0889581a)

tigrouind, 2024-09-20:
If anybody want to build this from the sources, here is the respective fixes : 
- [undo_redo_fix1](https://github.com/tigrouind/LDmicro/commit/a4661d007a87fdff9519d6f1c5c1f8e120158abf)
- [undo_redo_fix2](https://github.com/tigrouind/LDmicro/commit/73b910b3abf2f9120cfdb1c62ecd157106785681)